CoretechR/SPINC #1: Where do I get to buy this?

oz123, 2024-11-14:
I have only one thing to say:
Shut up and take my money!

Starry0Wolf, 2024-11-14:
Please, this is the best design the internet has seen for a battery charger!

lerra, 2024-11-15:
I was thinking the same thing, I would be happy to buy a pre built kit

CoretechR, 2024-11-15:
Thanks, I'm glad you like the project! I don't have plans for selling this at the moment. 
For my previous project, I wrote a Wiki page on how to order PCBs:
https://github.com/CoretechR/OMOTE/wiki/How-to-order-assembled-PCBs
LCSC should have all the electronical components except for the display and the DS2712. Those would have to be purchased from someone like Digikey. 
There is also a commercial charger that works very similarly: https://www.olightstore.com/ostation-x-super-charger

Starry0Wolf, 2024-11-15:
Oooo, that's some money. You know what, it would be fun to try to build this. Is there a AA version and a AAA version, or does it do both?

limited660, 2024-12-03:
Is there a BOM and CPL available for this?

CoretechR, 2024-12-03:
@limited660 I just added some files: https://github.com/CoretechR/SPINC/tree/main/PCB/SPINC%20AA%20Charger/production
I have not yet linked the JLCPCB part numbers as that's a manual process.

FrankGBusch, 2024-12-05:
I also would like to day "take my monnnneeeeeyy", but you already mentioned you're not going to sell this awesome device. So I'm about to build it by myself with the details your provided ... although I does not have any experience with PCBs. but I started to configure this thing at JLCPCB and it ends up with an overall cost of about $150 for five boards .. which is not bad from my POV. But really I'm not sure if I defined everything correctly so may I ask you to have a look at my order, just to verify everything's correct? Maybe via PM?

Thanks in advance
Frank

CoretechR, 2024-12-08:
> I also would like to day "take my monnnneeeeeyy", but you already mentioned you're not going to sell this awesome device. So I'm about to build it by myself with the details your provided ... [...]

@FrankGBusch Hi Frank, yes, I can take a look at your order.

SpaceBug199, 2024-12-20:
@FrankGBusch 
I have plenty of experience with electronics products and I have built a couple.
I can manage the PCB ordering and all the parts, testing and assembly. 

I'm thinking of ordering a batch of the PCBs and the components.

If there is interest I can do a batch of 5-10 pcs, let me know!
you can also email: qspectrumq -at- gmail.com
It would be the cost of the parts, 3D printer filament and shipping, I have no intent of profit.

Current batch -> 1/5 as of Dec 20 2024

akudlacek, 2024-12-20:
> @FrankGBusch I have plenty of experience with electronics products and I have built a couple. I can manage the PCB ordering and all the parts, testing and assembly.
>
> I'm thinking of ordering a batch [...]

Sign me up! I'll send you an email.

SpaceBug199, 2025-01-06:
@FrankGBusch, @oz123, @lerra , @Starry0Wolf 
I got a couple of people together to create the order.
If you are still interested you can email me qspectrumq@gmail.com

Starry0Wolf, 2025-07-17:
I do not have the perms to close issues, but this one should be closed.