CoretechR/SPINC #3: PCB manufacturer asking for updated BOM

abot7, 2025-01-26:
I tried to order the pcb with assembly from an online pcb maker, I'm not sure what I did wrong but their reply is below. Any idea how I can fix it?

"Our team reviewed your files for the “AA Charger | Rev. 2” project and need some clarification – can you please advise on the comments below?

 

Part C15849 is repeated in the bom for locations C1, C3 ,C15, C17, C2, C20, C4, C6, C8, C9,C30,C7.Should we combine them ?
Part C114626 is repeated in the bom for locations R3, R4,R5,R6.Should we combine them ?
Missing Mfgpartnum and partnum for location C10, C13, C14, C16, C18, C19, C21, C22, C24, C26, C27, C28
Missing Mfgpartnum and partnum for location C25, C5
Missing Mfgpartnum and partnum for location C29
Missing Mfgpartnum and partnum for location F1
Missing Mfgpartnum and partnum for location J2
Part DMG2301L twice in the bom for location Q1, Q3 and Q4.Should we combine them ?
Missing Mfgpartnum and partnum for location R15, R16, R19, R24, R25, R26, R27, R31, R36, R9
Missing Mfgpartnum and partnum for location R18, R23
Missing Mfgpartnum and partnum for location  R2
Missing Mfgpartnum and partnum for location R20
Missing Mfgpartnum and partnum for location  R21, R22, R28, R29
Missing Mfgpartnum and partnum for location R30
Missing Mfgpartnum and partnum for location R32
Missing Mfgpartnum and partnum for location R33
Missing Mfgpartnum and partnum for location R34
Missing Mfgpartnum and partnum for location R35
Missing Mfgpartnum and partnum for location  R37
Missing Mfgpartnum and partnum for location R7, R8
Can we get intelligent data prefer IPC-2581, ODB++ or ASCII CAD data.
 

An updated BOM would be preferred."

CoretechR, 2025-01-27:
Hi, the LCSC part numbers were indeed wrong. I'm very sorry about that!
The BOM should be fixed now (commit 9fcd008) but please verify the correct part assignment if you can.
And also be aware, that the DS2712 is not available from LCSC/JLCPCB.

tawsenior, 2025-01-27:
Who are you and why am I on this email chain?

*Troy A. Wilson Sr.*
847-275-8660 (Cell and GV)
***@***.***