CoretechR/SPINC #4: Finished circuit and not recognized by windows when plugged in

Raedrus, 2025-03-05:
Hi, i recently finished soldering the components onto the PCB, which i ordered from JLCPCB. However, when i plug the board into my PC, the LED lights up but the rp2040 is not detected. I did hold SW2 before plugging it in. I've also check every pin of all the chips and theres no bridging solder.

Here is a picture of my pcb

[image]

CoretechR, 2025-03-05:
Hi, the PCB looks OK apart from some solder blobs. Could you check if the 3.3V and 1.1V are present on the RP2040?
I sill would not rule out some connection problem. Normally, the RP2040 should show up on the PC right away.

Raedrus, 2025-03-05:
> Hi, the PCB looks OK apart from some solder blobs. Could you check if the 3.3V and 1.1V are present on the RP2040? I sill would not rule out some connection problem. Normally, the RP2040 should show up on the PC right away.

Hello thanks for the reply, Yes i did check the power, i forgot to mention it, the rp2040 is receiving the necessary power. I am thinking that i might reflow it once more, since the pins are so small there might be some disconnection at some pins.

CoretechR, 2025-03-05:
Yes, that might be a good idea. Any maybe try to clean the PCB with some isopropyl alcohol afterwards. 
I had to rework the RP2040 with a soldering iron, too.

Raedrus, 2025-04-02:
> Yes, that might be a good idea. Any maybe try to clean the PCB with some isopropyl alcohol afterwards. I had to rework the RP2040 with a soldering iron, too.

Tried and still no response in the PC lol, and burnt one led XD. I am thinking i should get new RP2040 chip and start over on a new pcb but that would be very frustrating.

CoretechR, 2025-04-03:
Sorry to hear that. Maybe it really just is an issue with the soldering and you need to assemble a new board. I don't have other ideas.